pedrolamas/klipper-preprocessor #1: Possible to add total filament usage to script?

PhilBaz, 2023-02-23:
### Is your feature request related to a problem? Please describe

Yes, when wanting to switch a spool that will be running out it is nice to be able to plan the pause for a number of reasons.  Knowing when to pause requires (for me) knowing how much of the cumulative filament has been used up, at each layer. Currently Cura doesn't provide this info.

### Describe the solution you'd like

A simple comment at the end of each layer saying how much cumulative filament has been used. The comment should be unique so it is searchable.

### Describe alternatives you've considered

Adding it with a spread sheet. echh.

### Additional information

The new Pause @ layer functionality is fantastic.  I was discussing with the developers at mainsail if it was possible to delay the pause until the infill section of that layer.  

Please see the brief discussion.
https://github.com/mainsail-crew/mainsail/issues/1283

Maybe it would be a possible option that could be integrated into the preprocessor rather than klipper/mainsail.

Thanks for your work!!!

pedrolamas, 2023-02-24:
Hi @PhilBaz thank you for opening this Feature Request.

> A simple comment at the end of each layer saying how much cumulative filament has been used. The comment should be unique so it is searchable.

This is not something that is currently possible. Unfortunately, Cura does not provide any data like that to the processing script (in fact, it doesn't even provide the layer info, but the script can still infer that from the Cura injected comments).

PhilBaz, 2023-02-24:
@pedrolamas 

Thank you Pedro.  Hopefully Cura will include this info at some point.

In terms of the layer number. Since it is being used to drive pause @ layer. Would it be possible to have the option to inject the layer change at the infill section of the layer? I know Cura also includes comments as to inner wall, outer wall, skin, infill.

Since pause often leaves artifacts on external surfaces, having the option to place the layer change (ultimately driving the pause)  at the infill portion of the layer would be ideal. 

Thanks again for your work Pedro.

pedrolamas, 2023-02-24:
Yes, it is possible (at least for Cura, can't say for others!), but that will require changes not only on this script, but also on Klipper, and Mainsail & Fluidd.

For Klipper, we would need a new parameter on `SET_PRINT_STATS` to pass the section type and then Klipper would store that and pass it on the status object (as it currently does for the total layer count and current layer index).

For this script, to inject a new `SET_PRINT_STATS` command when a new section starts.

For Mainsail & Fluidd, changes on these interfaces to make use of this new value.

PhilBaz, 2023-02-24:
@pedrolamas 

I mentioned this idea first to the Mainsail dev team. They said they would integrate this type of function if it was possible to identify the correct location. Ill link this thread in the discussion if you don't mind?

PhilBaz, 2023-03-01:
Hi Pedro,

I made a post on the Klipper forum in hopes that Kevin O'Conner will consider implementing the parameter you proposed that would allow for pause at infill.

Here is the post. If I have not described it properly can you please let me know if I should change anything. Thanks.

https://klipper.discourse.group/t/fr-new-parameter-on-set-print-stats-for-section-type-infill-wall-skin-etc/6943

theophile, 2023-03-02:
Came here from the Discourse thread. I'm a SuperSlicer/PrusaSlicer user, but if I'm understanding this request properly, I think the desired functionality is achievable. I'm currently fine tuning a perl script that is designed to minimize wasted filament at material changes by rearranging the sequence in which "features" are printed around toolchanges, so that "invisible" features (e.g. infill and supports) are grouped together and printed after the toolchange. The script also calculates the sum total of filament that will be extruded during these "invisible" features and passes that value to the toolchange macro, which then deducts that amount from the "purge" volume that would otherwise be used.  

I think you could do something similar with perl, python, or really any scripting language. It could read the gcode file down and parse all G1 commands to get the "E" value, then add that to the running total. At each layer change (or whenever you want) could have it insert a comment/macro/whatever that prints the current cumulative total. You could even have two totals running at the same time so you could track total filament used since print_start separately from total filament used this layer. 

If you wanted to be _really_ fancy you could read the gcode file into an array/list, have it sum up the total filament used for each layer, and then insert info at the desired locations to report total filament used since print start, used this layer, used next layer, used between now and print_end, etc.

PhilBaz, 2023-03-05:
@theophile 

Hi 

I'm not a programmer my self.  But I know the different feature types are given the the gcode of Cura as comments. Perhaps I can send you a sliced file to see if the script you are developing for Prusa/SS might also accommodate Cura as well?

Both the running filament count and pause @ inflll would be very useful for myself and I'm sure many many other Cura users.

closed 2023-12-21 by PhilBaz