pedrolamas/klipper-preprocessor #2: klipper-preprocessor not working with Smart Avionics Cura Master Branch

PhilBaz, 2023-02-24:
### What happened

The script does not appear to have any effect when used with smart avionics fork of Cura.  I tend to prefer his fork for its rock solid reliability and extra features. 

https://github.com/smartavionics/Cura

### What did you expect to happen

 For the script to work as in Cura 5.2/3.

### How to reproduce

Configure script and slice. 

### Additional information

Ill make a corresponding report at his git.

Thanks Pedro.

pedrolamas, 2023-02-24:
Can you please upload a sample gcode file sliced on this slicer so I can look and see what is going on?

I'm happy to fix this if it is a simple change that doesn't conflict with mainstream Cura, but I'm not very inclined to support unknown Cura forks as these can diverge a lot from what Cura provides.

PhilBaz, 2023-02-24:
@pedrolamas 

Fair enough.  Smart Avionics fork is a fairly significant fork. He is a very active developer and very responsive to community feedback with issues. I would imagine his fork is the most actively used of any.  He has developed many features that are excellent, his version is also totally rock solid compared to the UM 5 version.

I've included an Image of the settings and a sliced file.

[image]
[Untitled test.zip](https://github.com/pedrolamas/klipper-preprocessor/files/10826143/Untitled.test.zip)

PhilBaz, 2023-02-24:
[cura.log.1.zip](https://github.com/pedrolamas/klipper-preprocessor/files/10826263/cura.log.1.zip)

I searched for  ' klipper-preprocessor', but didn't get a hit. I'm not sure what else to look for. I attached the log.

Thanks Pedro.

pedrolamas, 2023-02-24:
This is the most recent entry on the log for scrips loading:

[image]

There is no indication there that KlipperPreprocessor was loaded (in fact, no reference anywhere on the log when I search for "KlipperPreprocessor")

cura.log.1 is rolled log, you should also have a cura.log which is the one we need here!

PhilBaz, 2023-02-24:
Ahh. here it is. I just sliced it again and saw the file date update immediately so hopefully something relevant should be in there.
[cura.zip](https://github.com/pedrolamas/klipper-preprocessor/files/10826401/cura.zip)

pedrolamas, 2023-02-24:
Here it is:

[image]

I have no idea where that `%1` bit came from...

pedrolamas, 2023-02-24:
What did you set for "Path to klipper_estimator"?

closed 2023-02-24 by PhilBaz