pedrolamas/klipper-preprocessor #4: klipper-preprocessor does not appear in "add a script" list

MonkeyMagic77, 2023-04-11:
### What happened

Placed the file in the script folder but get nothing in the add a script list in Cura 5.3.0

### What did you expect to happen

Expected the script to appear in the post processing script list as per instructions.

### How to reproduce

install cura 5.3.0 add KlipperPreprocessor.py to scripts folder.

pedrolamas, 2023-04-11:
Hi @MonkeyMagic77 what Operating System are you using? What is the full path to the script in your system?

MonkeyMagic77, 2023-04-11:
Hello, Im using windows 10 and C:\Users\George\AppData\Roaming\cura\5.3\scripts is the default Cura scripts folder.

pedrolamas, 2023-04-11:
Thanks, that seems correct (the full path should be something like "C:\Users\George\AppData\Roaming\cura\5.3\scripts\KlipperPreprocessor.py" for it to be found by Cura).

On the folder above you will find a "cura.log" can you upload that here please?

MonkeyMagic77, 2023-04-11:
[cura.log](https://github.com/pedrolamas/klipper-preprocessor/files/11200619/cura.log)

pedrolamas, 2023-04-11:
I think you might have downloaded the GitHub HTML page of the script, not the script contents!

[image]

pedrolamas, 2023-04-11:
Perfect, given that, I am closing this ticket now!