pedrolamas/klipper-preprocessor #12: Object cancelation doesn't properly recognize and label groups.

PhilBaz, 2023-12-21:
### What happened

I have some groups/assemblies. If I want to cancel one ideally it should show as a single 'object' for cancelation, but this doesn't happen. With 4 groups I end up with 7 objects. Also, none of the objects takes the name of the group, instead they object names are based on the name of the file, making it even harder to identify what I would like to cancel. Finally, the groups all show as overlapping polygons in mainsail making it impossible to identify and cancel the relevant group/object. 

### What did you expect to happen

Objects based on grouping, with the object name matching that of the group in the cura object list. 

### How to reproduce

Group a few things, slice and cancel.

### Additional information

Hi Pedro, I know your script only routes to the the cancelation preprocessor. But I thought maybe I should raise the issue with you first. I imagine this is likely an issue that must be fixed with the cancelation preprocessor exe?

pedrolamas, 2023-12-21:
Hi @PhilBaz, as you very well put, this issue is outside of the control of this script and you will need to raise it directly with the preprocessor cancellation repository.

PhilBaz, 2023-12-22:
@pedrolamas 

Thanks Pedro. I linked to an issue on the cancleation.exe git. Hopefully they will see it.

pedrolamas, 2023-12-22:
Following up on our discussion, I am now closing this issue.

closed 2023-12-22 by pedrolamas