pedrolamas/klipper-preprocessor #13: Bed Cooldown towards the end of a print

PatrickGlatz, 2024-01-22:
### Is your feature request related to a problem? Please describe

In OctoPrint i loved to cooldown the bed towards the end of a print to 
1. Save Energy
2. Make it easier to detach your print from the build plate (as mine is always super sticky when hot)

This is the link to the OctoPrint plugin, which allows for the same
https://plugins.octoprint.org/plugins/bedcooldown/

There it is solved by monitoring the print, but as klipper_estimator creates such perfect estimations, one would know that layer beforehand.

### Describe the solution you'd like

Identical to the OctoPrint Plugin, it would be perfect to have two additional fields for the configuration
1. Print time left threshold [s] (Default = 300)
2. Print completion threshold [%] (Default = 90)
3. Cooldown temp [°C] (Default = 0)

After the klipper_estimator has run, the script knows the exact time which a print durates. Based on this time, the script would insert 
`M140 S<COOLDOWNTEMP>`
when **both** conditions are met. The question is how to get the right position (maybe sum up times ?) or maybe cooperate with the maker of klipper_estimator? One would need a table with layer_num;progress;elapsed_time. Based on this and the total print duration the correct layer_num to insert can be identified and the GCODE can be implemented.

### Describe alternatives you've considered

1. slice Model in Cura
2. see resulting layer count
3. Calculate layer where to insert the bed cooldown, e.g. 45
4. Insert Post Processing Plugin 'Search and Replace'
    Search:     ;LAYER:(45)
    Replace:   ;LAYER:(45)\nM140 S0

### Additional information

I would REALLY REALLY love this (and my friends as well)

pedrolamas, 2024-01-23:
Hi @PatrickGlatz, thanks for the suggestion!

I can see value in your proposition, no question there, however I don't think this is in scope to the work done by this script.

The objective of this script is to be an intermediary between the g-code output of Cura slicer and Klipper, allowing to easily run other tools (such as klipper estimator) and add extra tags that are used in Klipper (such as the timelapse macros).

The feature you are describing can be implemented as a separate plugin, as there is value in that for any firmware - not just Klipper!

In terms of behavior of the existing OctoPrint plugin, seems to me that it monitors the print in real-time and then turns off the bed when the progress reaches a certain threshold. This can probably be done with a Klipper macro that continuously runs (delayed macro) and monitors the print progress.

closed 2024-01-23 by pedrolamas

PatrickGlatz, 2024-01-27:
@pedrolamas You are right, it is indeed better placed directly in the firmware (e.g. i am using mainsail / moonraker / klipper). As i am not too good in coding, i have decided to NOT write a moonraker plugin (although this would be the perfect solution and my mainbody is already finished, but i am not able to start the script when the print is starting, thats too sophisticated for me ^^. I really need somebody who holds my hand in this).

Instead i have taken your idea and had a look into delayed gcode. Fantastic solution and is working perfectly for me. I still have to write the Readme but maybe you want to have a glimpse:
https://github.com/PatrickGlatz/KlipperBedCooldown