pedrolamas/klipper-preprocessor #15: Specify API key

sesse, 2024-07-24:
### Is your feature request related to a problem? Please describe

Hi,

I'm running my network as if it is untrusted (i.e., I don't trust anything just because it comes from a local IP address, and I run TLS on everything); however, then Klipper preprocessing refuses to run; on upload from Cura, I get this:

```
Failed to run klipper_estimator
Failed to load printer configuration: Access denied (you may need to use --config_moonraker_api_key): HTTP status client error (401 Unauthorized) for url (https://<my URL here>/printer/objects/query?configfile=settings)
```

But there's no way in the Cura UI to specify that API key.

### Describe the solution you'd like

A text box in the UI where I can enter a Moonraker API key; if nonempty, should be given to klipper_estimator.

sesse, 2024-07-24:
Perhaps something like this? (Rather untested)

```diff
@@ -114,6 +114,14 @@
                     "default_value": "",
                     "enabled": "klipper_estimator_enabled and klipper_estimator_config_type == 'moonraker_url'"
                 },
+                "klipper_estimator_moonraker_api_key":
+                {
+                    "label": "Moonraker API key",
+                    "description": "API key to access Moonraker (only needed if on an untrusted network)",
+                    "type": "str",
+                    "default_value": "",
+                    "enabled": "klipper_estimator_enabled and klipper_estimator_config_type == 'moonraker_url'"
+                },
                 "klipper_estimator_config_cache": {
                     "label": "Cache config from Moonraker",
                     "description": "Enable this to cache a copy of the config from Moonraker.",
@@ -231,23 +239,26 @@
         if klipper_estimator_enabled:
             klipper_estimator_config_type: str = self.getSettingValueByKey("klipper_estimator_config_type")
             klipper_estimator_moonraker_url: str = self.getSettingValueByKey("klipper_estimator_moonraker_url")
+            klipper_estimator_moonraker_api_key: str = self.getSettingValueByKey("klipper_estimator_moonraker_key")
             klipper_estimator_config_file_path: str = self.getSettingValueByKey("klipper_estimator_config_file_path")
             klipper_estimator_path: str = self.getSettingValueByKey("klipper_estimator_path")
 
+            args = [
+                klipper_estimator_path,
+            ]
+            if klipper_estimator_config_type == 'moonraker_url':
+                args += ['--config_moonraker_url', klipper_estimator_moonraker_url]
+                if klipper_estimator_moonraker_api_key:
+                    args += ["--config_moonraker_api_key", klipper_estimator_moonraker_api_key]
+            else:
+                args += ['--config_file', klipper_estimator_config_file_path]
+
             if klipper_estimator_config_type == 'moonraker_url' and self.getSettingValueByKey("klipper_estimator_config_cache"):
                 Logger.log("d", "Running klipper_estimator to get config from Moonraker...")
 
-                klipper_estimator_config_type = 'file'
-
-                args = [
-                    klipper_estimator_path,
-                    "--config_moonraker_url",
-                    klipper_estimator_moonraker_url,
-                    "dump-config"
-                ]
-
                 config_filename = os.path.join(work_dir, "config.json")
 
+                args += ['dump-config']
                 with open(config_filename, 'w') as config_file:
                     try:
                         ret = subprocess.run(args, stdout = config_file, startupinfo = self.getSubprocessStartupinfo(), timeout = 5)
@@ -256,14 +267,14 @@
                     except subprocess.TimeoutExpired:
                         pass
 
-            Logger.log("d", "Running klipper_estimator...")
-
-            klipper_estimator_config_arg: str = klipper_estimator_moonraker_url if klipper_estimator_config_type == 'moonraker_url' else klipper_estimator_config_file_path
+                args = [
+                    klipper_estimator_path,
+                    '--config_file',
+                    config_filename
+                ]
 
-            args = [
-                klipper_estimator_path,
-                "--config_" + klipper_estimator_config_type,
-                klipper_estimator_config_arg,
+            Logger.log("d", "Running klipper_estimator...")
+            args += [
                 "post-process",
                 filename,
             ]
```

sesse, 2024-07-24:
Thanks, that was quick! I'll give it a shot once this print completes. :-)

sesse, 2024-07-24:
I'm afraid I get exactly the same error message. From looking at the code, it seems you don't handle the case of not caching the config? So the --config_moonraker_api_key flag is never given, since I never take the dumpconfig path.

pedrolamas, 2024-07-24:
> I'm afraid I get exactly the same error message. From looking at the code, it seems you don't handle the case of not caching the config? So the --config_moonraker_api_key flag is never given, since I never take the dumpconfig path.

Well spotted, I've re-opened the ticket and will add the code for the other case too!

pedrolamas, 2024-07-24:
And now fixed on https://github.com/pedrolamas/klipper-preprocessor/releases/tag/v1.4.1

closed 2024-07-24 by pedrolamas