pedrolamas/klipper-preprocessor #16: klipper-preprocessor does not appear in "add a script" list - Cura 4.7.1

bengalih, 2025-04-08:
### What happened

I don't know if Cura 4.7.1 is not supported.  I don't see any information on the main page about supported Cura versions.
Assuming it is supported, the script is not appearing.

Cura.log states:

`2025-04-08 17:01:42,534 - ERROR - [MainThread] UM.Logger.logException [106]: Exception: Exception occurred while loading post processing plugin: invalid syntax (KlipperPreprocessor.py, line 160)
2025-04-08 17:01:42,538 - ERROR - [MainThread] UM.Logger.logException [110]: Traceback (most recent call last):
2025-04-08 17:01:42,542 - ERROR - [MainThread] UM.Logger.logException [110]:   File "C:\Program Files\Ultimaker Cura 4.7\plugins\PostProcessingPlugin\PostProcessingPlugin.py", line 197, in loadScripts
2025-04-08 17:01:42,545 - ERROR - [MainThread] UM.Logger.logException [110]:     spec.loader.exec_module(loaded_script)  # type: ignore
2025-04-08 17:01:42,549 - ERROR - [MainThread] UM.Logger.logException [110]:   File "<frozen importlib._bootstrap_external>", line 693, in exec_module
2025-04-08 17:01:42,553 - ERROR - [MainThread] UM.Logger.logException [110]:   File "<frozen importlib._bootstrap_external>", line 799, in get_code
2025-04-08 17:01:42,556 - ERROR - [MainThread] UM.Logger.logException [110]:   File "<frozen importlib._bootstrap_external>", line 759, in source_to_code
2025-04-08 17:01:42,560 - ERROR - [MainThread] UM.Logger.logException [110]:   File "<frozen importlib._bootstrap>", line 222, in _call_with_frames_removed
2025-04-08 17:01:42,564 - ERROR - [MainThread] UM.Logger.logException [110]:   File "C:\Users\bhendin\AppData\Roaming\cura\4.7\scripts\KlipperPreprocessor.py", line 160
2025-04-08 17:01:42,567 - ERROR - [MainThread] UM.Logger.logException [110]:     add_timelapse_take_frame: bool = self.getSettingValueByKey("add_timelapse_take_frame")
2025-04-08 17:01:42,571 - ERROR - [MainThread] UM.Logger.logException [110]:                             ^
2025-04-08 17:01:42,575 - ERROR - [MainThread] UM.Logger.logException [110]: SyntaxError: invalid syntax`

This is on a Windows 11 system.

I do plan to upgrade Cura, but right now I am comfortable on 4.7.1 and want to work out all the kinks in my current migration to Klipper first before  I throw a new verison in the mix.  If the issue is with 4.7.1, I will just wait until I'm able to upgrade.

In the meantime, is there any way to just get Mainsail's pause at layer functions working via Cura?

Thanks.

### What did you expect to happen

Klipper script appears

### How to reproduce

Place python script in scripts directory and restart Cura.

pedrolamas, 2025-04-09:
Hi @bengalih thank you for opening this issue.

I have tested the script with Cura 4.13 and it worked fine there, so I would expect it to work with any Cura 4.x though I have not tested it there.

I would check if the script file downloaded correctly (try to open the file on Notepad or any other text editor and compare the contents [with this](https://raw.githubusercontent.com/pedrolamas/klipper-preprocessor/refs/heads/master/KlipperPreprocessor.py))

bengalih, 2025-04-09:
It appears that the python version must have been updated to 3.6 post Cura 4.7 and so it does not appear to support the `: type` annotation syntax.

I went through the code and made the following changes:

`$ diff -u orig.py KlipperPreprocessor.py | grep -E '^[-+]'
--- orig.py     2025-04-09 14:33:53.068112000 -0500
+++ KlipperPreprocessor.py      2025-04-09 14:24:31.327573000 -0500
-        add_timelapse_take_frame: bool = self.getSettingValueByKey("add_timelapse_take_frame")
+        add_timelapse_take_frame = self.getSettingValueByKey("add_timelapse_take_frame")  # Removed : bool
-        add_set_print_stats_info: bool = self.getSettingValueByKey("add_set_print_stats_info")
+        add_set_print_stats_info = self.getSettingValueByKey("add_set_print_stats_info")
-        data: List[str] = []
+        data = []  # Removed : List[str]
-            layer: List[str] = []
+            layer = []
-        preprocess_cancellation_enabled: bool = self.getSettingValueByKey("preprocess_cancellation_enabled")
+        preprocess_cancellation_enabled = self.getSettingValueByKey("preprocess_cancellation_enabled")
-            preprocess_cancellation_path: str = self.getSettingValueByKey("preprocess_cancellation_path")
+            preprocess_cancellation_path = self.getSettingValueByKey("preprocess_cancellation_path")
-            timeout: int = self.getSettingValueByKey("preprocess_cancellation_timeout")
+            timeout = self.getSettingValueByKey("preprocess_cancellation_timeout")
-        klipper_estimator_enabled: bool = self.getSettingValueByKey("klipper_estimator_enabled")
+        klipper_estimator_enabled = self.getSettingValueByKey("klipper_estimator_enabled")
-            klipper_estimator_config_type: str = self.getSettingValueByKey("klipper_estimator_config_type")
-            klipper_estimator_moonraker_url: str = self.getSettingValueByKey("klipper_estimator_moonraker_url")
-            klipper_estimator_moonraker_api_key: str = self.getSettingValueByKey("klipper_estimator_moonraker_api_key")
-            klipper_estimator_config_file_path: str = self.getSettingValueByKey("klipper_estimator_config_file_path")
-            klipper_estimator_path: str = self.getSettingValueByKey("klipper_estimator_path")
+            klipper_estimator_config_type = self.getSettingValueByKey("klipper_estimator_config_type")
+            klipper_estimator_moonraker_url = self.getSettingValueByKey("klipper_estimator_moonraker_url")
+            klipper_estimator_moonraker_api_key = self.getSettingValueByKey("klipper_estimator_moonraker_api_key")
+            klipper_estimator_config_file_path = self.getSettingValueByKey("klipper_estimator_config_file_path")
+            klipper_estimator_path = self.getSettingValueByKey("klipper_estimator_path")
-            klipper_estimator_config_arg: str = klipper_estimator_moonraker_url if klipper_estimator_config_type == 'moonraker_url' else klipper_estimator_config_file_path
+            klipper_estimator_config_arg = klipper_estimator_moonraker_url if klipper_estimator_config_type == 'moonraker_url' else klipper_estimator_config_file_path
-            timeout: int = self.getSettingValueByKey("klipper_estimator_timeout")
+            timeout = self.getSettingValueByKey("klipper_estimator_timeout")`

This appears to have allowed the extenstion to load without error, but I have not fully tested yet.